edwardtfn/TX-Ultimate-Easy #9: BLE support

edwardtfn, 2024-05-25:
Request 7:
Ble support?

*Suggested by @TeDeVPrime on #2*

edwardtfn, 2024-05-29:
This one is gonna be challenging. I really wanna use BT on that device, as I believe it's signal may be much better than a NSPanel, however the i2c audio component used for Media Player is only available for Arduino at this moment.
I will keep this open as I still wanna investigate it more, but will certainly be out of the initial release.

TeDeVPrime, 2024-05-29:
the git i gave you from the other project isn't it based on ESP?
the speaker works there as a media player

edwardtfn, 2024-05-29:
No, those are all Arduino.

edwardtfn, 2024-05-29:
This is what happens with ESP-IDF:

```log
Failed config

light.neopixelbus: [source /config/esphome/packages/edwardtfn/TX-Ultimate-Easy/ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml:292]
  
  This feature is only available with frameworks ['arduino'].
  id: light_full
  name: Light - All
  internal: True
  platform: neopixelbus
  type: GRB
  variant: WS2811
  num_leds: 32
  pin: GPIO13
media_player.i2s_audio: [source /config/esphome/packages/edwardtfn/TX-Ultimate-Easy/ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml:1503]
  
  This feature is only available with frameworks ['arduino'].
  id: mp_speaker
  name: Speaker
  platform: i2s_audio
  i2s_dout_pin: GPIO15
  i2s_audio_id: if_i2s_audio
  i2s_comm_fmt: lsb
  dac_type: external
  mode: mono
  mute_pin: 
    number: GPIO26
    inverted: True
    allow_other_uses: True
  internal: False
  ```
  
The light I'm pretty confident is just about finding another driver available, but the `media_player.i2s_audio` will require some review in the component in ESPHome. 😞 

Anyhow, I could change the flash size to 8MB and with this I can build including the BLE with no size error, however it is crashing on boot, so I will have to investigate it more.

edwardtfn, 2024-06-12:
Monitoring #6894 as that could enable the use of Media Player and NeoPixBus with ESP-IDF, making some memory available for BLE.

BartoszPierzchlewiczMacrix, 2025-03-03:
Hi, I think `bluetooth_proxy` component is not working in the current state of the repo despite the fact, that `ESPHome/TX-Ultimate-Easy-ESPHome_addon_ble_proxy.yaml` is disabling meda_player.
Uncommenting `ESPHome/TX-Ultimate-Easy-ESPHome_addon_ble_proxy.yaml` causes that TX-Ultimate has boot loop.

closed 2025-08-01 by edwardtfn