edwardtfn/TX-Ultimate-Easy #12: Media Player - Improve sound volume

edwardtfn, 2024-05-29:
Hello again, it's me. Spiderman.

I am pretty sure that you will include the speaker in the integration but i am just mentioning it just in case.

the most common problem to everyone flashing the TX Ultimate to this day is the sound volume.
With the original firmware everything is fine and has a satisfactory sound level.

With every flashed firmware, the sound is extremely low. If you are 1 meter away... you can't hear anything.

Maybe you can figure it out in the process?

*This is a split of #10, reported by @TeDeVPrime.

TeDeVPrime, 2024-06-04:
how is the project going? i tried to install it but i got stuck the other day. it couldn't communicate

edwardtfn, 2024-06-05:
> how is the project going? 

I had to stop a bit during this week, due to other things, but I have a working version where you can control the relays. Quite simple right now, and basically using the very same components as used in other implementations, so with similar results, including the media player probably as loud as in other implementations.
Still pending the cover support, which I'm testing first on the NSPanel project, then I will just copy it to TX Ultimate.
I wanna have the basics working first, then I will try to go deeper on those.

> i tried to install it but i got stuck the other day. it couldn't communicate

It woukd be really nice to have other eyes on it and start getting some feedback. If you can give more details on what got you stuck I'd love to invest some time on it.
Currently I'm trying to take advantage of the 8MB flash available, but as you have another firmware installed mapped to 4MB you may have issues with OTA. You can either flash via serial or enforce 4MB (which is totally workable right now).