edwardtfn/TX-Ultimate-Easy #13: led color sync

mon3y78, 2024-06-08:
It would be nice to be able to choose for the LED coloring to be synchronized with the color of the lamp light when it is turned on.

mon3y78, 2024-06-11:
Now that you have added the ability to disable the toggle, it would be very nice to synchronize the color/state of the button's LED with the light. Don't you think it would be convenient to create virtual buttons based on the chose model?

edwardtfn, 2024-06-11:
> Don't you think it would be convenient to create virtual buttons based on the chose model?

Could you please give more detail about what is in your mind. 😉

mon3y78, 2024-06-12:
- The Wi-Fi/Zigbee lights must always remain powered. The button action line needs to include the state on (on/toggle/off).
[image]
- The LED lighting of the button area must reflect the state (on/off) of the light/entity. Using MQTT, managing the state should not be an issue.
[image]

mon3y78, 2024-06-12:
Hello, if I can help you in any way, just let me know!

edwardtfn, 2024-12-19:
You mean having an option for the Button x always turn on Relay x or always turn off Relay x?
Or do you just wanna set one light to be always On or Off? If so, why not controlling the light directly and disabling the respective button?

> The Wi-Fi/Zigbee lights must always remain powered

What do you mean by this? Are you using the lights on the TX Ultimate to show Wi-Fi status? Do you wanna this to be local?

edwardtfn, 2024-12-23:
Well, if you are controlling a light without using the relays, you have an external automation for that and the TX Ultimate isn't aware of which entity is connected to, so it can't adjust itself to that entity state.
This could be changed, of course, but we have to even do this in an automation (Blueprint) or find a way to inform the panel the `entity_id` of the light being controlled by a button. Unfortunately ESPHome doesn't have an entity picker, so again a Blueprint could be an option, or entering this as a substitution on yaml, which is easier to implement, but less user's friendly.
I will take a look at this later. I don't exclude the possibility of a very simple Blueprint, but that adds quite a lot of complexity to this project, as keeping the two sides updated to the same version is much harder than just updating ESPHome, so I will look at the substitutions as a first solution to be improved later. 😉 
No promises here. 😉

mon3y78, 2024-12-26:
I use [](https://github.com/alexdelprete/ha-blueprints)