edwardtfn/TX-Ultimate-Easy #14: Swipe not working properly.

TeDeVPrime, 2024-06-09:
Hello,

i managed to install it finally on my TX Ultimate 3 relay.

The issues i have noticed so far is that if i swipe left or right.... the sensors don't get updated and they stay ON.

the worst issue with it is that if i touch the switch and try to swipe... the relays always activate. 
so if relay 1 turns the light on.... and swipe up opens the cover....
when i try to open the cover, the light also turns on.

so the event of the relay action must only happen on the release and not press

edwardtfn, 2024-06-09:
> so the event of the relay action must only happen on the release and not press

Good point... I will look at that.

edwardtfn, 2024-06-11:
I believe this is fixed in `dev`. I will move it to `main` later today.
Could you please give it a try and let me know?

edwardtfn, 2024-06-11:
I'm reopening this as I could see the issue persists.

TeDeVPrime, 2024-06-22:
any luck with this?

edwardtfn, 2024-06-22:
I have some ideas for this, but couldn't work with this project recently. The latest ESPHome came with some break changes that got my attention, now I'm getting prepared to move to a new house...
But I will try to look at thin in the coming days.

TeDeVPrime, 2024-06-23:
the latest ESPHome broke your house and now you need to move to a new one? i should stay away from that update then.

but no worries :) let me know when you have time again. enjoy your new home

TeDeVPrime, 2024-07-19:
Hi Edward,

did you find time to check this out?
@edwardtfn

edwardtfn, 2024-07-22:
I couldn't work at this yet, but I promise to do it later this week.
Sorry for that.

edwardtfn, 2024-07-26:
I've finally assembled by device back, but could run only very limited tests.
Could you please give it a try and let me know the results?

TeDeVPrime, 2024-12-21:
this is not fixed. if you try to slide...
the first button that you touch while going to slide... it gets activated

closed 2024-12-22 by edwardtfn

edwardtfn, 2024-12-22:
Could you please try again? 😬

cavazosomar, 2024-12-22:
Hi, on a 3 gang US model swiping up or down appears as swipe left or right on the logs, therefore the corresponding sensors never turn on

edwardtfn, 2024-12-22:
Well, technically, the US model is a portrait like device... When you rotate it to be landscape, the device isn't aware of that, so, from the point of view of the device, you are swiping to the side.
If that is an issue for you, please add a feature request for a "US landscaped" mode, but I'm not gonna make any promises here, as covering all the possible positions on both models will be a nightmare.
What I can do (in fact this was already started) is moving all those strings to substitutions, so you can set it yourself in your local yaml and it will use the labels you want.

cavazosomar, 2024-12-22:
What I meant to say is that the swipe actions don´t appear to do anything, either in HA or in the web page of the device. Since I saw the id definitions for all swipe directions on TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml I thought that the swipe events were mapped differently for the EU and US versions with the recent update

[image]
[image]
[image]

edwardtfn, 2024-12-22:
Got it!
Let's work with this in the new issue:
- #51
I will try to work at this later today.