edwardtfn/TX-Ultimate-Easy #16: Controlling the Leds for each side.

TeDeVPrime, 2024-06-09:
Hello again,

I saw that there is no option of controlling the leds for each side. 
Yes i can have the lights top and bottom for when the relays are active 
but i want to control the left and right side as two different lights with different led effects.

this is how i will be able to have a visible notification for when my alarm is set.
i will set the right side as Red Pulse Slow with 20% brightness

edwardtfn, 2024-06-11:
Left and right lights are published, but are disabled by default.
Could you please double check the "entities not shown"?

TeDeVPrime, 2024-06-11:
yes you are right. i didn't see them.
they look great!

one issue that i have discovered is that if i choose red pulse slow 100% it plays just fine... but if i choose 20% it will change for 1 second and then the effect will go back to 100% (even though home assistant says it is at 20%)

edwardtfn, 2024-06-12:
This is the default behavior of pulse effect in ESPHome: https://esphome.io/components/light/index.html#pulse-effect
It's probably possible to change this at runtime, but it will be a bit complex. Better would be creating a [feature request on ESPHome](https://github.com/esphome/feature-requests/issues) to limit the max brightness on pulse to the light brightness.

closed 2024-06-12 by edwardtfn