edwardtfn/TX-Ultimate-Easy #20: Not ideal led numbers for activation of Relay 1 and Relay 3

TeDeVPrime, 2024-06-11:
I did some tests and the leds you activate at the bottom (and top) for relay 1 and relay 3 are not the best choice.

at the moment it seems that you activate 7 for relay 3, when it should be 6
and 11 for relay 1 instead of 12 (which should be)

the reason i am saying this is because if you have Relay 1 or 3 ON, you are not sure if relay 2 is on since they are stuck together.
if Relay 2 is on, you are not sure if 1 or 3 is on as well.

setting the 6 and 12, it makes a better distinguish between them.
the same goes for the top leds (i can give you the led numbers if you need)

edwardtfn, 2024-06-11:
I can use the respective lights on the top.
Thanks.

edwardtfn, 2024-06-12:
I will use this picture as a reference:

[image]

Currently, when using a 3-gang device, the following LEDs are currently in use:

| Button/Relay | LED<br>Bottom | LED<br>Top |
| :--: | :--: | :--: |
| 1 | 12 | 22 |
| 2 | 10 | 24 |
| 3 | 8 | 26 |

I will change to the following:
| Button/Relay | LED<br>Bottom | LED<br>Top |
| :--: | :--: | :--: |
| 1 | 13 | 21 |
| 2 | 10 | 24 |
| 3 | 7 | 27 |

edwardtfn, 2024-06-12:
Done!

closed 2024-06-12 by edwardtfn