edwardtfn/TX-Ultimate-Easy #22: Request: Long Press on each button?

TeDeVPrime, 2024-06-11:
Is it me or do i saw a double click / long click event in your yaml?

From the sensors i can see that you know if the button is being pressed or not so i am guessing it can be done?

edwardtfn, 2024-06-11:
Yes, I currently have events for click, double click and long click, but as I've set those as normal binary sensors we can use any of the automations available: https://esphome.io/components/binary_sensor/index.html#binary-sensor-automation

I have, however, to solve the issue with a click being detected on swipe or multi-fingers touch.

edwardtfn, 2024-06-11:
By the way, the click, double-click and long-press events are sent to Home Assistant as events. You can monitor those on your Developer Tools / Events area and can use in your automations.

edwardtfn, 2024-06-11:
I'm not sure I can detect a long multi-touch. Will have to look deeper on this.
Most likely we can.

Shach01, 2024-06-18:
Hi Edward, thank you very much for the work. I am trying to decrypt the serial between the ESP32 and the touch controller. do you need to send a JSon or initilisation to the controller on boot? we are getting a constant generic stream over serial that does not change with touch inputs. any guidance would be much appreciated. Have a great day