edwardtfn/TX-Ultimate-Easy #23: Default actions if the switch is offline or HA is offline

TeDeVPrime, 2024-06-11:
i am not sure if it works like that since i can't test it at the moment but...

if the switch can't connect to HA, can we have a backup that it works as a normal switch? meaning that relays work on / off when you press them?
even if you have disabled the button action in HA or the touch panel power?

edwardtfn, 2024-06-11:
Currently everything happens locally independently of wifi or HA, so if a button is assigned to a relay, it will toggle the relay anyways.
Is this what you are asking for or are you asking for an alternative action to be executed only when offline? Like a fallback action per button?

TeDeVPrime, 2024-06-12:
no i want a fallback action so that if the TX cannot connect to HA,
to work as a regular relay switch in case that i have disconnected the button from the relay.

this is low priority. we are just throwing ideas here

edwardtfn, 2024-12-21:
So, coming back to this...
Currently, there are user's selectable actions for buttons, which can be either "None" or "Relay x (toggle)".
What you are asking for is a third option where it will do nothing, unless the network or Home Assistant are out, when it should fall back to the "Relay x (toggle)" mode.
Is that correct?

TeDeVPrime, 2024-12-21:
YOU ARE ALIVE!!!! i thought the house ate you :D

yes this is exactly what i want. 
For example.... the left button toggles the WiFi bulb to turn on or off (without cutting the power) so... None.

but if HASS dies... or network dies.... the switch needs to work as a proper switch (Relay x toggle)

that's correct.

edwardtfn, 2024-12-21:
> i thought the house ate you :D

It almost... lots of things to fix and while I'm fixing something, lots of new issues starts popping up around me... 😞 😆

edwardtfn, 2024-12-23:
@TeDeVPrime, I cannot test this right now, so if you have a change to try, please set to `ref: button-failsafe-action` and flash it into your panel (wirelessly is fine).
Please let me know the results of your tests as I will wait for that to merge it.

TeDeVPrime, 2024-12-23:
@edwardtfn so... i flashed it like you said. 
then i changed the settings to Relay 1 (API failsafe only) and restarted the switch.

then i unplugged my home assistant from the network and tried the switch.

all it does is vibrate on my presses. probably it thinks that there is a connection to HASS?

edwardtfn, 2024-12-23:
It can be that your device is taking longer to detect the API outage.
Please try adding this to your device's yaml temporarily. It will blink a green light every time an API connection happens, and red on each disconnection. Remember the logger itself counts as an API connection, so it may connect/disconnect without loosing connection to API.
If you have no API connections at all, it will be solid RED. Then try the fallback mode at that time.
If that works, then it will be just about improving the API outage detection.

```yaml
api:
  on_client_connected:
    then:
      - light.turn_on:
          id: light_00
          brightness: 100%
          red: 0
          green: 100%
          blue: 0
      - delay: 500ms
      - if:
          condition:
            - api.connected:
          then:
            - light.turn_off:
                id: light_00
  on_client_disconnected:
    then:
      - light.turn_on:
          id: light_00
          brightness: 100%
          red: 100%
          green: 0
          blue: 0
      - delay: 500ms
      - if:
          condition:
            - api.connected:
          then:
            - light.turn_off:
                id: light_00
```

TeDeVPrime, 2024-12-24:
i installed the changes.... restarted the device.... unplugged my HASS.... i saw no lights (green or red)
and MAYBE after 5 minutes it seemed to work for just a few seconds.... then it went unresponsive again. it just vibrates when i touch it and does nothing...

edwardtfn, 2024-12-24:
I'm trying to find out how long ESPHome will take to detect the API disconnected... looks like there's a 60s keepalive, but also a 2,5min ping timeout, which I think it is what is in use... I have to investigate it more.
Anyways, I don't understand why your panel isn't turning red with the code above. Please try adding this one to your WiFi (and keep the one above related to api):
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  on_connect:
    then:
      - light.turn_on:
          id: light_13
          brightness: 100%
          red: 0
          green: 0
          blue: 100%
      - delay: 500ms
      - if:
          condition:
            - wifi.connected:
          then:
            - light.turn_off:
                id: light_13
  on_disconnect:
    then:
      - light.turn_on:
          id: light_13
          brightness: 100%
          red: 100%
          green: 100%
          blue: 0
      - delay: 500ms
      - if:
          condition:
            - wifi.connected:
          then:
            - light.turn_off:
                id: light_13
```

edwardtfn, 2024-12-24:
I've also changed the code to test WiFi also, and not only API state. Maybe that will be detected earlier.

closed 2024-12-24 by edwardtfn