edwardtfn/TX-Ultimate-Easy #24: Help with Serial Reads from touch controller.

Shach01, 2024-06-18:
Hi Adward, we are struggling to get usable serial information from the touch controller. it is always the same stream no matter if we attemp to trigger any form of touch event:

AA 55 1 7 0 62 3B 46 57 53 57 45 53 2D 54 35 54 4F 55 43 48 2D 54 35 2D 43 

any help or guidance would be much appreciated, have a great day man!

Kind regards,

Josh

edwardtfn, 2024-07-12:
Hey @Shach01, did you got what you were looking for?

Shach01, 2024-07-12:
Hey Edward!

We did cone right! After reviewing the drivers for tasmota and esp easy we realized (not documented) that they are pulling GPIO5 HIGH to activate the touch driver, weird as the touch driver still spams a default serial stream and once you pull this pin high the driver behaves and starts to send correct touch serial data.

Thank you so much for your assistance. I hope that helps.

Kind regards,

Josh

closed 2024-12-19 by edwardtfn