edwardtfn/TX-Ultimate-Easy #27: Refactor suggestion - Splitting into multiple files (packages pattern)

ROBOT0-VT, 2024-07-28:
Right now, I can only imagine that working with a single 2000-line file with the base config cannot be great for maintainability. I suggest splitting the `yaml` config using esphome's `packages` feature, to make this easier to work with.

A good example of this (and where I learned this) is [Karl Quinsland's TX Ultimate firmware](https://karlquinsland.com/sonoff-t5-esphome/), although there does not appear to be an easily-accessible repo for this example 😭

edwardtfn, 2024-12-19:
Solved in DEV and will be part of the coming release.
It can be improved in the future, but a first step is done.

edwardtfn, 2024-12-20:
Could you please update to the latest version and try again?
Please let me know the results.

closed 2024-12-20 by edwardtfn