edwardtfn/TX-Ultimate-Easy #28: After the last changes the Relay button lights no longer work

TeDeVPrime, 2024-08-03:
@edwardtfn 
I did the new update and the swipe seems to be working fine now 
but i lost the lights when the relay is on.
am i doing something wrong? here is a screenshot

[image]

edwardtfn, 2024-08-03:
I will take a look

edwardtfn, 2024-12-20:
Could you please update to the latest version and try again?
Please let me know the results.

TeDeVPrime, 2024-12-21:
@edwardtfn nope. it doesn't work. If i set my middle relay to Light as shown in the picture above... it will not turn on the light.

it is like the switch is attached to the relay mode and not the state of the button. 
if i turn the Relay 2 Mode back to a Relay.... then the lights on the switch come on

iceteaSA, 2024-12-21:
@edwardtfn On my side (US-Variant 2-4 Gang) it is mostly working correctly, except only the right side LEDs are mapped correctly, the left side seems to be off, with the top LEDs coming on for the top gangs. Another issue is that state is not restored properly on boot. The status lights will still burn, and HA reports the lights to be on, but the relays are not triggered, meaning the actual lights are still off. Also, the LEDs reset back to white on boot if you have changed it.

closed 2024-12-23 by edwardtfn

edwardtfn, 2024-12-23:
This should be fixed now. Could you please update your system and try it again?