edwardtfn/TX-Ultimate-Easy #29: some problem with  TX Ultimate 4 Gang US (T5-4-120)

mon3y78, 2024-08-14:
Hi, I'm having some issues with the TX Ultimate 4 Gang US (T5-4-120):

If I turn on the light, I'm unable to turn it off again.
The same light doesn't fully illuminate (photo attached).
All the LEDs (top-right-left-bottom) are not correctly synchronized (I’m attaching a photo of the top light), and they don't turn off.
Have other people had the same problems?
I can't figure out how to correct the error!
[image]
[image]

edwardtfn, 2024-08-15:
I can try to,duplicate the issues, however, I’m unsure if I fully understood all of those. Could you please provide more details of what was expected behaviour and how it is behaving for each of the issues you reported?

mon3y78, 2024-08-15:
[image]

mon3y78, 2024-08-28:
🤔

edwardtfn, 2024-08-28:
Sorry, I was quite busy those days...
I will look at this later today.

mon3y78, 2024-08-30:
Unfortunately, the problem hasn't been resolved. A short section of the LEDs is not lighting up.
Additionally, when I turn on the LEDs, it's not possible to turn them off. The switch on the panel turns off, but the LEDs remain on.
Is there anything I can do to help you understand the issues?
Thank you for your efforts.

edwardtfn, 2024-12-20:
Could you please update to the latest version and try again?
Please let me know the results.

closed 2024-12-20 by edwardtfn