edwardtfn/TX-Ultimate-Easy #48: Light's Going Off

iceteaSA, 2024-12-21:


edwardtfn, 2024-12-21:
By default, ESPHome restarts after 15min without API connection (Home Assistant connection):
> reboot_timeout (Optional, [Time](https://esphome.io/guides/configuration-types#config-time)): The amount of time to wait before rebooting when no client connects to the API. This is needed because sometimes the low level ESP functions report that the ESP is connected to the network, when in fact it is not - only a full reboot fixes it. Can be disabled by setting this to 0s. Defaults to 15min.

Source: https://esphome.io/components/api.html#configuration-variables

As the light state is not saved (yet, as a solution for #47 may change that), it will be off after the restart.

You can prevent the restart by setting this `reboot_timeout` to a higher time or to 0s to disable it.
Just add this to your panel's yaml:
```
api:
  reboot_timeout: 24h  # or 0s if you wanna disable this feature completely
```

iceteaSA, 2024-12-21:
@edwardtfn Thanks! I'll change that

I'm still pretty new to esphome, and HA, so sorry for the rookie mistake!

closed 2024-12-21 by iceteaSA