edwardtfn/TX-Ultimate-Easy #51: were mapped differently for the E update

edwardtfn, 2024-12-22:
What I meant to say is that the swipe actions don´t appear to do anything, either in HA or in the web page of the device. Since I saw the id definitions for all swipe directions on TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml I thought that the swipe events were mapped differently for the EU and US versions with the recent update

[image]
[image]
[image]

_Originally posted by @cavazosomar in https://github.com/edwardtfn/TX-Ultimate-Easy/issues/14#issuecomment-2558326991_

edwardtfn, 2024-12-22:
@TeDeVPrime, are you still having this issue with the latest version? I cannot duplicate this anymore.

TeDeVPrime, 2024-12-22:
@edwardtfn yes these are my settings
[image]

i have None in the button actions since i want to control other devices and i don't want to hear the annoying relay noises all the time.

the moment i touch the switch.... the sensors on Button 1 - 2 - 3 activate and the swipe action as well.

My automations are based on the button sensors

edwardtfn, 2024-12-22:
You are running v0.0.4. The version now is something like v2024.12.x.
Try cleaning build files and installing again.

TeDeVPrime, 2024-12-22:
@edwardtfn i did install with main instead of dev.

same thing is happening. the Sensors catch the event when i place my finger and then the swipe...

[image]

edwardtfn, 2024-12-22:
Just to align the expectations... This is how I'm expecting the system to work:

When you touch the panel, regardless if you swipe or not, the binary sensor will turn on and will stay on until you remove your finger from the panel, when the binary sensor will then turn off.
The binary sensor isn't directly connected to the click action, however.
When a click is detected (you put your finger and remove within 350ms without changing position), it will execute the command defined in the button action selector.
If you swipe, the swipe sensor will register that for the time defined in your config.
Any click, double click or long press will be send to Home Assistant as events.

TeDeVPrime, 2024-12-22:
if i set the button 1 action to None (like i have done...) it will not register any actions... so i have to base my events on the sensors.

i reason i was forced to use the sensors and not the actual button is because if i set it to Relay instead of none.... it triggers the relays and i don't want that.

i don't even understand what's the difference between Relay 1 mode = Light and Button 1 Action = None

so ok. let's say i am using it wrong and i agree.... how can i achieve my goal of executing things without triggering the actual relays?

edwardtfn, 2024-12-22:
> if i set the button 1 action to None (like i have done...) it will not register any actions... so i have to base my events on the sensors.

Well, not really.
In Home Assistant, sensors (including binary sensors) are design to represent states, not events. And a button can have two states, pressed or not pressed, and this is what those sensors are representing.
For events, Home Assistant offers the "[Events](https://www.home-assistant.io/docs/configuration/events/)", and this is what we are using. A button click (press and then release in a short time) is a event, a swipe is another event, and a long press is another event. Events are probably the most low level and reliable thing in Home Assistant, as you can rely on unique ids, etc. The event can contain lots of info.
So, my recommendation is that you use [events for triggering](https://www.home-assistant.io/docs/automation/trigger/#event-trigger) your automations.
If you wanna know how an event looks like, go to Developer Tools, on the tab "Events", type `esphome.ts_ultimate_easy` in the "Event to subscribe to" field and click "Start listening", then play with your panels and you will see each event there.

[image]

We can add any information needed on the events, so just let me know if you miss some info there or if you want an information in a different format and I can look at that.

By the way, in your automations, you will have the variable `trigger.event`, so if you are using yaml to adjust your automation, you can have something like this:
```yaml
- if:
    - '{{ trigger.event.component == "bs_button_3" }}'
    - '{{ trigger.event.event == "long_click" }}'
  then:
    - do_something
```

You can, or course, use the UI for Event based triggers. Just in the event data area, device_id to filter your panel and the fields you wanna to check for your trigger.
Please let me know if you need more details or some examples. I will try to work on some documentation around this.

 
> i reason i was forced to use the sensors and not the actual button is because if i set it to Relay instead of none.... it triggers the relays and i don't want that.
> 
> i don't even understand what's the difference between Relay 1 mode = Light 

This will define if your TS's relay will be represented as a switch or as a light on the "Controls" area on your device's page (under Settings > Devices & services > ESPHome).

> and Button 1 Action = None

This indicates your TX Ultimate will do nothing locally when Button 1 is pressed. You could chose it to toggle the Relay 1, which would be executed regardless if that is set as a Light or a Switch, and regardless if the Wi-Fi or Home Assistant is available to your TX Ultimate.

> so ok. let's say i am using it wrong and i agree.... how can i achieve my goal of executing things without triggering the actual relays?

I believe I've responded this, but please let me know if your still need more details.

TeDeVPrime, 2024-12-22:
can't i just have Button 1 - 2 -3 actions have another option in there instead of relay called "Soft Event" (for example) which doesn't trigger the relay when pressed?
why make life hard when all i want is a simple node-red event on the button press that doesn't trigger the relay?

edwardtfn, 2024-12-22:
I will add the "Relay x (API failsafe)" option as requested in #23, but right now, if you select "Button x action" to "None", the relays aren't triggered and you still getting the button click events in Home Assistant to use in your automations. 😉

edwardtfn, 2024-12-23:
So, for now, I've added new sensors for the events, so in top of the events sent to Home Assistant, you will also have sensors for "Button 1 - Click event", "Button 2 - Double-click event", "Button 3 - Long-press event", etc.
I will publish this very soon.

edwardtfn, 2024-12-23:
Done!
Could you please update your system and try it again?

closed 2024-12-23 by edwardtfn

mon3y78, 2024-12-23:
Amazing!

mon3y78, 2024-12-23:
Witch 2024.12.9 I can't disable button action

TeDeVPrime, 2024-12-23:
ok with the new version....

if you swipe once.... the switch crashes....

so.... i have my button 3 on...
i try to swipe left.... the command executes just fine and then.... it dies... the leds turn off.... commands on presses don't work but the touch vibration is working.... after a few seconds (like 10) the switch works again.

if you try to swipe right.... same things happens

cavazosomar, 2024-12-26:
Swipping for me now works however multitouch event doesn't update (it is registered in the logs however)
[image]
[image]