edwardtfn/TX-Ultimate-Easy #52: Speaker not working in the last update?

TeDeVPrime, 2024-12-22:
I can't seem to make any sound come out of the device. i am using google say to make it say "Hi" or something.... nothing is happening. I get no errors in hasss

edwardtfn, 2024-12-23:
I could duplicate this. In fact, I got some memory allocation messages and then a crash on the device:
```log
[04:45:39][D][esp-idf:000]: E (841672) I2S: i2s_alloc_dma_buffer(712): Error malloc dma buffer

[04:45:39][D][esp-idf:000]: E (841675) I2S: i2s_realloc_dma_buffer(759): Failed to allocate dma buffer

[04:45:39][D][esp-idf:000]: E (841687) I2S: i2s_set_clk(1687): I2S0 tx DMA buffer malloc failed

[04:45:39][D][esp-idf:000]: E (841699) I2S: i2s_driver_install(1985): I2S set clock failed
```

I will look at this tomorrow.

edwardtfn, 2024-12-23:
The code is exactly the same as the one used back in July. It could be that other components are taking more memory, but it's hard to believe the system got out of memory for this.
Looks like something broken on ESPHome side. I will have to test it with a legacy version and take a look on their changes around these components.

edwardtfn, 2024-12-23:
- esphome/issues#6497

edwardtfn, 2024-12-23:
No, it isn't fixed yet. 😩 
I haven't tried the adding the delay yet, but I will look a bit more for a permanent solution before using a working around.

edwardtfn, 2024-12-23:
Tests results: (in progress)
 | ESPHome version | Results |
| :--: | :--: |
| 2024.12.2 | Crashed |
| 2024.10.3 | Crashed |
| 2024.8.3 | Crashed |
| 2024.7.3 | Testing... |

edwardtfn, 2024-12-23:
It is failing even with legacy versions, so I have to investigate if HA have changed the call since July (when it was supposed to work fine).

TeDeVPrime, 2024-12-23:
i appreciate the time you are giving this

mon3y78, 2024-12-27:
After update 2024.12.19 -> 2024.12.24 my speaker don't work[image]

mon3y78, 2024-12-27:
Same for vibrate

edwardtfn, 2024-12-27:
> After update 2024.12.19 -> 2024.12.24 my speaker and vibrate don't work! I use us 3gang [image](https://github.com/user-attachments/assets/750e8b90-788b-4c92-ad22-9bbefbc715de)

Was the Media Player working for you in v2024.12.19?? 😲 
Do you mind sharing your device's yaml? Please remove any sensitive info before sharing.

mon3y78, 2024-12-27:
Maybe I’m wrong… I have several TX Ultimate at home. How can I reinstall version 19?
My YAML is the standard one… I have no issues sharing logs or files if needed.
Also, since the latest versions, I no longer have vibration when pressing the buttons.

edwardtfn, 2024-12-27:
> How can I reinstall version 19?

Change `ref: main` to `ref: v2024.12.19` and flash it again.

edwardtfn, 2024-12-27:
> My YAML is the standard one… I have no issues sharing logs or files if needed. Also, since the latest versions, I no longer have vibration when pressing the buttons.

If you running v2024.12.22 or later, your yaml, change file name from `TX-Ultimate-Easy-ESPHome_core.yaml` to `TX-Ultimate-Easy-ESPHome.yaml`, but this won't work with v2024.12.19, where you should leave as it is (with `core` in the name.

mon3y78, 2024-12-27:
With 19 i can see mediaplayer
[image]
But if i try to send a text it reboot

[image]

edwardtfn, 2024-12-27:
Ok, this is the same issue I have here.
I've intentionally removed the Media Player on v2024.12.22 as it was crashing. I'm still working at this to have it back, but for now, I've replaced it by a `speaker` component, which is available now on the latest version, but don't show an entity in the UI.
I will return with the Media Player as soon I can find a solution for this.
I've also break the system into `core`, `standard` and `addon`. This was the intentional since the beginning, but I couldn't have time to work at this before. The `core` package includes only the very minimum for your panel to work, so `vibration`, `relays` and `speaker` are not there. Those all have being moved to `standard` package, so in order to have those, you must either add `TX-Ultimate-Easy-ESPHome_standard.yaml` or remove `core` from your file name to access a file which includes both `core` and `standard`.
Please use the example in the repo to review your device's yaml: https://github.com/edwardtfn/TX-Ultimate-Easy/blob/main/TX-Ultimate-Easy-ESPHome.yaml

closed 2025-01-06 by edwardtfn