edwardtfn/TX-Ultimate-Easy #57: Request: ability to set the color and brightness of status lights from the ui

cavazosomar, 2024-12-23:
Currently the status lights are only white at 100% brightness. With this enhancement the color and brightness status lights should be able to be configured directly in the ui

edwardtfn, 2024-12-25:
I've just added an option to expose the relay indicator's lights to Home Assistant. With that, you will be able to set the settings for those lights, which will be used when the respective relay turns on.
Please update your system and let me know if this is ok.
I will look later for a way to set how those lights should be when the relay is off.

closed 2024-12-25 by edwardtfn

cavazosomar, 2024-12-26:
The button to expose the relay indicators didn't do anything for me, however the entities did appear on the disabled list. I just had to enable them manually

edwardtfn, 2024-12-26:
And are those working properly after you enable it manually?

cavazosomar, 2024-12-27:
So I checked again after updating to the latest version and I can see that the entities are getting exposed ( and hidden) when toggling the switch, however they are added to the entities list as disabled every time either the toggle switch is pressed or when updating the firmware even if I enabled them previously, so they have to be enabled every time. I think it would be better if the were added as enabled by default

edwardtfn, 2024-12-27:
WIth the latest changes those entities shouldn't be disabled when exposed:
https://github.com/edwardtfn/TX-Ultimate-Easy/pull/77/files

So either you're not with the latest build (which right now is [![Version][version-shield]](https://github.com/edwardtfn/TX-Ultimate-Easy/tags) or your system is maintaining the old settings in flash (and I'm not sure what can be done in this case other than a full reset which will discard all other settings saved 😞 ).

It's also important to note that some changes requires your system to be restarted before those become effective. I've added a sensor named **Pending Restart Status** on the **Diagnostic** area which should indicate a problem when your system is pending restart to make some change valid.

[version-shield]: https://img.shields.io/github/v/tag/edwardtfn/TX-Ultimate-Easy?label=version
[version]: https://github.com/edwardtfn/TX-Ultimate-Easy/tags

cavazosomar, 2024-12-30:
Well I just updated to v2024.12.27 and the entities still appear as disabled, as seen in the screenshot: 

[image]

I suppose it's not that much of a hassle to add those entities manually from time to time once the code become more mature and the updates become less frequent

PR #71 (2024-12-25): Expose relay-indicator LEDs
The lights will be available on Home Assistant, so users will be able to set it's color and brightness, then this will be used when the relay turn on-off as the indicator for the relay status.
Resolves #57

## Summary by CodeRabbit

- **New Features**
	- Introduced a new binary sensor, `bs_pending_restart`, to track pending restart status.
	- Added multiple new light entities for relay configurations.
	- Changed the designation of a component from "Speaker" to "Media Player."

- **Bug Fixes**
	- Enhanced handling of gang count during initialization, improving state  [... cut]
commit 403cdce2 (2024-12-25): Expose relay-indicator LEDs

The lights will be available on Home Assistant, so users will be able to set it's color and brightness, then this will be used when the relay turn on-off as the indicator for the relay status.
Resolves #57