edwardtfn/TX-Ultimate-Easy #65: if you swipe once.... the switch crashes....

edwardtfn, 2024-12-23:
ok with the new version....

if you swipe once.... the switch crashes....

so.... i have my button 3 on...
i try to swipe left.... the command executes just fine and then.... it dies... the leds turn off.... commands on presses don't work but the touch vibration is working.... after a few seconds (like 10) the switch works again.

if you try to swipe right.... same things happens

_Originally posted by @TeDeVPrime in https://github.com/edwardtfn/TX-Ultimate-Easy/issues/51#issuecomment-2560428552_

edwardtfn, 2024-12-23:
@TeDeVPrime, which version are you running?
I'm trying to duplicate this, but it is working fine on my system (using US model also). I have no crashes on swipe...

TeDeVPrime, 2024-12-23:
@edwardtfn i have the EU version and my version is 2024.12.12

the swipe moves my cover up or down

edwardtfn, 2024-12-23:
Is it crashing with swipe?
I have a EU model for testing, but I have to set it up... will do that tomorrow.

TeDeVPrime, 2024-12-23:
@edwardtfn 

here is a video
https://www.youtube.com/watch?v=Fq9A-ukvSlk&ab_channel=tedev

edwardtfn, 2024-12-24:
I couldn't duplicate with EU model also.
Could you please share some logs and your device's yaml? Please hide any sensitive info before sharing here.

TeDeVPrime, 2025-01-04:
hi @edwardtfn , i found a forgotten node red automation that when i used swipe up or down, i was sending the tts to the switch to say that the covers are coming down or going up. i am guessing that the speakers was forcing the switch to crash. so this is not a swipe issue. it is a speaker issue

edwardtfn, 2025-08-02:
By the way, was your goal to have a sound feedback on swiping?

closed 2025-08-05 by edwardtfn