edwardtfn/TX-Ultimate-Easy #67: alternative led configuration us

mon3y78, 2024-12-24:
for us 3 gan:
  - platform: partition
    name: "LEDs Button right"
    id: leds_button_right
    internal: true
    segments:
      - id: leds
        from: 6
        to: 7      
      - id: leds
        from: 16
        to: 17
  - platform: partition
    name: "LEDs Button middle"
    id: leds_button_middle
    internal: true
    segments:
      - id: leds
        from: 3
        to: 4     
      - id: leds
        from: 19
        to: 20  
  - platform: partition
    name: "LEDs Button left"
    id: leds_button_left
    internal: true
    segments:
      - id: leds
        from: 0
        to: 1      
      - id: leds
        from: 22  
        to: 23
for 4 gang 4:
- platform: partition
    name: "LEDs Button l4"
    id: leds_button_l4
    internal: true
    segments:
      - id: leds
        from: 7
        to: 8
      - id: leds
        from: 15
        to: 16
  - platform: partition
    name: "LEDs Button l3"
    id: leds_button_l3
    internal: true
    segments:
      - id: leds
        from: 5
        to: 6
      - id: leds
        from: 17
        to: 18
  - platform: partition
    name: "LEDs Button l2"
    id: leds_button_l2
    internal: true
    segments:
      - id: leds
        from: 1
        to: 2
      - id: leds
        from: 21
        to: 22
  - platform: partition
    name: "LEDs Button l1"
    id: leds_button_l1
    internal: true
    segments:
      - id: leds
        from: 0
        to: 0
      - id: leds
        from: 23
        to: 24
      - id: leds
        from: 31
        to: 31   

[image]

edwardtfn, 2024-12-24:
On the 3-gang (US) version, the difference you are proposing it to use the LEDs 16 and 17 instead of 17 and 18 currently in use for Relay 3 - Left.

https://github.com/edwardtfn/TX-Ultimate-Easy/blob/ae87f025a33d7c036ec29895ac6e986da6a6576e/ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml#L642-L648

edwardtfn, 2024-12-24:
And for the 4-gang (US) version you are proposing to use 2-LED per relay instead of 1-LED per relay on the current implementation.

https://github.com/edwardtfn/TX-Ultimate-Easy/blob/ae87f025a33d7c036ec29895ac6e986da6a6576e/ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml#L656-L711

edwardtfn, 2024-12-24:
Ok, forget my comments about the 4-Gang model... After applying the changes I could see the issue with the previous mapping.

edwardtfn, 2024-12-24:
Could you please update your system and try again?

PR #68 (2024-12-24): New LED mapping for US 3&4-Gang
Partially resolves #67

## Summary by CodeRabbit

- **New Features**
	- Updated segment definitions for the light relay configuration, adjusting the range of controlled LEDs.

- **Bug Fixes**
	- Ensured the overall functionality and logic of the light relay system remain intact after modifications.
closed 2024-12-24 by edwardtfn