edwardtfn/TX-Ultimate-Easy #74: Multitouch event doesn't update

edwardtfn, 2024-12-26:
Swipping for me now works however multitouch event doesn't update (it is registered in the logs however)
[image]
[image]

_Originally posted by @cavazosomar in https://github.com/edwardtfn/TX-Ultimate-Easy/issues/51#issuecomment-2562120130_

Matthijs33, 2024-12-29:
@edwardtfn 

I am not sure if you already take an look in to this issue, but the issue has been found in "TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml" on rule 321, the state was here "OFF" instead of "ON":

```c#
  - id: touch_on_multi_touch_release
    mode: restart
    then:
      # Extended by:
      #   - HW Buttons
      #   - HW Vibration
      - binary_sensor.template.publish:
          id: bs_multi_touch
          state: ON  # <- This was OFF
```

edwardtfn, 2024-12-30:
Great that you found it.
I'm not at home today, but will apply the change as soon I can get something better than my phone screen to edit it. Or feel free to submit a PR changing this. 😉

TeDeVPrime, 2025-01-06:
is this fixed? i tried to update but the multi touch is still not working

PR #90 (2025-01-06): Fix value for multi_touch event publishing
fixes #74
Value was "OFF" instead of "ON"

## Summary by CodeRabbit

- **New Features**
	- Enhanced touch panel responsiveness
	- Updated multi-touch and swipe event handling
- **Documentation**
	- Added comments about hardware button and vibration integration for touch events
commit 48ddfbf8 (2025-01-06): Update TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml

fixes #74
Value was "OFF" instead of "ON"
closed 2025-01-06 by edwardtfn