edwardtfn/TX-Ultimate-Easy #82: Firmware version changing on reboot?

cavazosomar, 2024-12-30:
After updating to the latest version (v2024.12.27 currently) I realized that just after finishing the update process a few features don't work such as the relays, vibration, configuration selections, and the device webpage, even the restart button doesn't do anything. After a power reboot the device does come online with everything working however the firmware version appears as v2024.12.19 instead of the version I just uploaded

Right after upload:
[image]
[image]
[image]

After reboot:
[image]

edwardtfn, 2024-12-30:
On your device's yaml, please change from `TX-Ultimate-Easy-ESPHome_core.yaml` to `TX-Ultimate-Easy-ESPHome.yaml` and flash it again. It should all be back.

cavazosomar, 2024-12-30:
Now the device doesn't require a reboot to start working howeverr I'm still getting the same results, the firmware still reports v2024.12.19 even though it displays v2024.12.27. 
I tried changing `ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml ` for `TX-Ultimate-Easy-ESPHome.yaml` in my yaml file and directly appending  `- ESPHome/TX-Ultimate-Easy-ESPHome_standard.yaml`    to my original file (which already included a reference to ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml). Both with the same results
     
I'm attaching the log from the last firmware update

[logs_tx-ultimate-3g_run.txt](https://github.com/user-attachments/files/18273218/logs_tx-ultimate-3g_run.txt)

timeline where the firmware version is mentioned:
[02:24:22]
[02:25:48]

arnonh, 2025-01-06:
I think it is some sort of fail-safe of the bootloader meaning the v2024.12.27 version doesn't work reliably ,
try testing with the new "revert-to-media-player-as-default" branch

edwardtfn, 2025-03-31:
Hey @cavazosomar, are you still having those issues or should we close this?

edwardtfn, 2025-07-30:
Should we close this?

closed 2025-07-31 by edwardtfn