edwardtfn/TX-Ultimate-Easy #89: Cover mode for blind control

brotaxt, 2025-01-05:
Hey guys, thank you very much for all your work! :) 

In the current documentation there is no mention of a ‘cover mode’ to control roller shutters. Is this already possible? 
As it only works with an interlock between two relays.

Cheers,
brotaxt

edwardtfn, 2025-01-05:
That isn't available yet, but shouldn't be hard to implement.
I will take a look, but as I cannot test it here, will need your support with some feedback when I have a testing code.

brotaxt, 2025-01-05:
That's no problem :) I have several switches here to test new code. Thank you very much! ♥️

TeDeVPrime, 2025-01-06:
i don't understand what the "cover mode" is. you can make it work as a cover controller through automations.

for example i just have an automation for the slide for my cover.
i have turned the switch 90 degrees to the right. so i slide down (right) if i want the cover to come down or up (left) if i want the cover to go up.

edwardtfn, 2025-01-06:
I think the basic difference would be the capability of controlling this locally even when Wi-Fi isn't available.

ShonP40, 2025-02-09:
Any updates on this?
I can help test it as well

edwardtfn, 2025-08-03:
So, I wanna look at this, finally, but I need some help to brainstorm how this will work.

Some things that come to my mind (please feel free to comment and add your thoughts):
1. It shouldn't support 1-Gang devices
2. Needs an interlock (this is on software level on ESPHome, but it should work fine)
3. It must support `cover` type entity published to Home Assistant via API
4. It will have to be time based, as there's no other way to know when cover is closed or open
    - It could be based on [ESPHome native Time Based Cover](https://esphome.io/components/cover/time_based)
    - Will require some time calibration or input on HA UI and a backup time via substitutions
        - could be a number selected by user (preferable) or some calibration procedure (less intuitive)
        - Must remember between boot
5. A button (touch area) could be set to either open and close, user defined
6. Relays for open and close should also be user defined

Anything else?

ShonP40, 2025-11-01:
Any updates on this?