edwardtfn/TX-Ultimate-Easy #93: Swipe up/down binary sensors missing

brotaxt, 2025-01-14:
I've tested different branches & tags but the binary sensor for swipe up/down does not show up in HA. 

[image]

edwardtfn, 2025-01-14:
The swipe mode is defined by the hardware. While the US model supports swipe left/right, the EU model will support up/down (or vice versa, I'm not sure right now).
Depending on your selection of model, it will enable the corresponding sensors.
But anyways, you cannot have both up/down and left/right in the same device, unfortunately. This is a hardware limitation.

brotaxt, 2025-01-14:
Oh okay, thank you for that clarification. I'm using the EU version and can only see swipe left and swipe right. I will close this issue then.

closed 2025-01-14 by brotaxt