edwardtfn/TX-Ultimate-Easy #95: Speaker issue? Am i doing something wrong? Loud static noise and then it freezes

TeDeVPrime, 2025-01-15:
So i updated to the latest and used core and standard for files.

then i tried to use google translate service to send a tts command
a loud static noise came from the speaker and then it rebooted.

here is the log file?

[23:15:19][W][component:237]: Component script took a long time for an operation (327 ms).
[23:15:19][W][component:238]: Components should block for at most 30 ms.
[23:15:49][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[23:15:49][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:15:49][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:15:51][D][light:036]: 'Light - LED 20' Setting:
[23:15:51][D][light:047]:   State: OFF
[23:15:51][D][light:085]:   Transition length: 0.5s
[23:15:51][D][light:036]: 'Light - LED 12' Setting:
[23:15:51][D][light:047]:   State: OFF
[23:15:51][D][light:085]:   Transition length: 0.5s
[23:16:12][D][media_player:065]: 'Media Player' - Setting
[23:16:12][D][media_player:072]:   Media URL: http://192.168.3.60:8123/api/tts_proxy/QXFpHYiD0iVnPGiT-hWs_Q.mp3
[23:16:12][D][media_player:078]:  Announcement: yes
[23:16:12][W][component:237]: Component i2s_audio.media_player took a long time for an operation (532 ms).
[23:16:12][W][component:238]: Components should block for at most 30 ms.
[23:16:14][W][component:237]: Component i2s_audio.media_player took a long time for an operation (341 ms).
[23:16:14][W][component:238]: Components should block for at most 30 ms.
[23:16:14][W][component:237]: Component i2s_audio.media_player took a long time for an operation (328 ms).
[23:16:14][W][component:238]: Components should block for at most 30 ms.
[23:16:16][D][media_player:065]: 'Media Player' - Setting
[23:16:16][D][media_player:072]:   Media URL: http://192.168.3.60:8123/api/tts_proxy/QXFpHYiD0iVnPGiT-hWs_Q.mp3
[23:16:16][D][media_player:078]:  Announcement: yes
[23:16:16][D][esp-idf:000]: E (87173) I2S: i2s_alloc_dma_buffer(712): Error malloc dma buffer

[23:16:16][D][esp-idf:000]: E (87175) I2S: i2s_realloc_dma_buffer(759): Failed to allocate dma buffer

[23:16:16][D][esp-idf:000]: E (87187) I2S: i2s_set_clk(1687): I2S0 tx DMA buffer malloc failed

bartoszpierzchlewicz, 2025-03-01:
I have the same, media player is not working

edememinov, 2025-03-03:
<html>
<body>

15:03:44 | [D] | [media_player:073] | 'Media Player' - Setting
-- | -- | -- | --
15:03:44 | [D] | [media_player:080] | Media URL: http://172.19.0.11:8123/media/media/dry-fart.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiIxYTVkNTI4NWUzODI0MTQyODQ2MjIwOWNjOGU2ZjgxMCIsInBhdGgiOiIvbWVkaWEvbWVkaWEvZHJ5LWZhcnQubXAzIiwicGFyYW1zIjpbXSwiaWF0IjoxNzQxMDEwNjI2LCJleHAiOjE3NDEwOTcwMjZ9.vfS9BwimsSWIscglrDbCW8xdpRl0R9bjemM3_7CT0IQ
15:03:44 | [D] | [esp-idf:000] | E (326138) I2S: i2s_alloc_dma_buffer(712): Error malloc dma buffer
15:03:44 | [D] | [esp-idf:000] | E (326140) I2S: i2s_realloc_dma_buffer(759): Failed to allocate dma buffer
15:03:44 | [D] | [esp-idf:000] | E (326152) I2S: i2s_set_clk(1687): I2S0 tx DMA buffer malloc failed
15:03:44 | [D] | [esp-idf:000] | E (326165) I2S: i2s_driver_install(1985): I2S set clock failed

</body>
</html>

I don't even get the loud static noise, it just crashes. Seems like an old issue, that has returned

BartoszPierzchlewiczMacrix, 2025-08-01:
I think the first time you change platform you should use cable (serial) to upload firmware, because of partitioning change.

edwardtfn, 2025-08-01:
Could you please share your yaml?

edememinov, 2025-08-01:
I replaced packages with the snippet you provided. I think I might be doing something wrong. But when I try to build, this is the error that I am getting:

```YAML
INFO ESPHome 2025.7.4
INFO Reading configuration /config/tx-ultimate-office-easy-office-test.yaml...
Failed config

light.esp32_rmt_led_strip: [source /data/packages/c8868aaa/ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml:47]
  id: light_full
  name: Light - All
  internal: True
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  
  [rmt_channel] is an invalid option for [light.esp32_rmt_led_strip]. Please check the indentation.
  rmt_channel: 1
  chipset: ws2812
  num_leds: 32
  pin: GPIO13
  restore_mode: RESTORE_DEFAULT_OFF
  max_refresh_rate: 33ms
  use_psram: True
```

I am on ESPHome version 2025.7.4 (in case that helps)

edwardtfn, 2025-08-01:
Please try changing `ref: main` to `ref: 2025.8.1` in order to force a reload.
Also, please change `refresh: 5min` to `refresh: 5s` to force it loading more frequently.

closed 2025-08-02 by edwardtfn