edwardtfn/TX-Ultimate-Easy #97: Incompatible with new ESPHome version (2025.2)

math77FR, 2025-02-19:
Hi,

Fyi, failed config with the new version of ESPHome (2025.2.0).
Could you please update packages ?

`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/kitchen-tx-ultimate.yaml...
INFO Updating https://github.com/edwardtfn/TX-Ultimate-Easy@stable
Failed config

light.esp32_rmt_led_strip: [source /data/packages/c8868aaa/ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml:47]
  id: light_full
  name: Light - All
  internal: True
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 1
  chipset: ws2812
  num_leds: 32
  pin: GPIO13
  restore_mode: RESTORE_DEFAULT_OFF
  max_refresh_rate: 33ms
  use_psram: True`

bartoszpierzchlewicz, 2025-02-24:
https://esphome.io/changelog/2025.2.0.html#removal-of-old-style-platform-configuration

bartoszpierzchlewicz, 2025-02-24:
... or maybe I'm wrong, and real cause is here: https://github.com/esphome/esphome/pull/7770

math77FR, 2025-02-26:
Thanks @bartoszpierzchlewicz 

Works correctly with ESPHome 2025.2.1

I have previously had major WIFI disconnection problems with the stable version. Hopefully this will be a thing of the past with Latest.

closed 2025-03-01 by math77FR