edwardtfn/TX-Ultimate-Easy #100: On_press = Turn on / On_release = Turn off Option

Telk0, 2025-03-11:
Hello,

Is it posible to add this function to the code? this way you will be able to control a dimmer with hold to dim.

Telk0, 2025-03-29:
Up? 

@Edwardtfn

edwardtfn, 2025-03-31:
I believe right now you should be able to see the state of the button (if pressed or not)? Isn't that working?

Telk0, 2025-03-31:
I need the relay to turn on with state pressed and turn off with release to be able to control dimmer Directly

Med venlig hilsen / Best regards
Martin Madsen

Telk0, 2025-05-17:
Status update @edwardtfn

edwardtfn, 2025-05-19:
My apologies for the delay here.
I cannot test this with my current setup, but please try adding this to your yaml temporarily and let me know if it works properly so I can add to the main code:

```yaml
binary_sensor:
  - id: !extend bs_button_1
    on_press:
      then:
        - switch.turn_on: sw_relay_1
    on_release:
      then:
        - switch.turn_off: sw_relay_1
```

Telk0, 2025-05-20:
@edwardtfn 

Im a little confused with the "advanced" yaml files you have made :)
Which to put this? you have all the "sub" yaml files for relays, buttons etc.

Using this currently:

substitutions:
  name: tx-ultimate-easy
  friendly_name: TX Ultimate Easy

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: latest  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    refresh: 5min
    files:
      # Core (essential) packages - Do not disable these unless you know what you're doing
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml            # Basic shared settings
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml        # Button logic
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml           # LED configuration
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml          # Touch panel support

      # Optional functionality - Can be enabled/disabled as needed
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml     # Relay control
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml  # Haptic feedback

      # Audio options (use none or choose only one - using both will fail)
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml  # Media player (Recommended for most users)
      # - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_speaker.yaml  # Basic speaker

Telk0, 2025-05-20:
@edwardtfn 

I have testet the new code.

Issues found:
On VERY fast touch,  the relay toggle on/off and then turn on and stay on. 
If you hold the button for a sec or more the turn on while holding and off on release works good.
Toggle does not work with swipe/multitouch.. if you try to swipe or multitouch, the first button you hit st in the action will toggle the relay. both Button 1 and Swipe left sensor activates. Not sure this can co-exist :)

edwardtfn, 2025-05-20:
Ok, I will have to work a bit more on this and disable other actions when this is enabled.

Telk0, 2025-05-20:
Maybe its posible to add an delay? That sensors first respond after certain ms?
Same with the quick touch issue that keep relay on? maybe an touch delay can help with this?

edwardtfn, 2025-05-20:
Maybe we can use the binary sensor for long press instead... I will look at that, hold on...

edwardtfn, 2025-05-20:
Please try this and let me know:

```yaml
binary_sensor:
  - id: !extend bs_button_1
    on_release:
      then:
        - switch.turn_off: sw_relay_1

  - id: !extend bs_button_1_long_press_event
    on_press:
      then:
        - switch.turn_on: sw_relay_1
    on_release:
      then:
        - switch.turn_off: sw_relay_1
 ```

Telk0, 2025-05-20:
This code is not working at all. Now there are no momentary at all or on_press hold.

edwardtfn, 2025-05-20:
Ok. I will have to investigate this deeper.

Telk0, 2025-06-08:
@edwardtfn had a chance to take a deeper look?

Telk0, 2025-06-16:
> Ok. I will have to investigate this deeper.

Any luck @edwardtfn

Telk0, 2025-09-07:
Had a chance to look at this @edwardtfn ?

Telk0, 2026-01-20:
Will test it when get home. When device reboot it goes back to default settings.. Toggle and light mode bottom..

Telk0, 2026-01-20:
Its set to "latest" version.
Doing factory_reset clears errors and accept all  settings afterwards.

adding below causes the relay to toggle off before i release the button.
 tx_ultimate_easy:
  on_long_touch_release:
    then:
      - switch.turn_off: sw_relay_1
  on_multi_touch_release:
    then:
      - switch.turn_off: sw_relay_1
  on_release:
    then:
      - switch.turn_off: sw_relay_1

edwardtfn, 2026-03-20:
So, you wanna a diferent behaviour between button 1 and other buttons?

closed 2026-10-01 by github-actions[bot]