edwardtfn/TX-Ultimate-Easy #102: Failed to load packages

chadwo, 2025-03-27:
Anxious to give this a try as I've already got ESPHome flashed and not much of a programmer but I'm getting an error on install "Failed to load packages" which seems to be related to the URLs it's trying to fetch. I used the advanced version:

INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/kitchen-tx.yaml...
Failed config

packages: [source /config/esphome/kitchen-tx.yaml:20]
  remote_package: 
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: stable
    refresh: 5min
    files: 
      
      Failed to load packages. ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml does not exist in repository.
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml

Failing from HA terminal but working if I add "/blob/main/" into URL:

       ▄██▄           _   _                                    
     ▄██████▄        | | | | ___  _ __ ___   ___               
   ▄████▀▀████▄      | |_| |/ _ \| '_ ` _ \ / _ \              
 ▄█████    █████▄    |  _  | (_) | | | | | |  __/              
▄██████▄  ▄██████▄   |_| |_|\___/|_| |_| |_|\___|          _   
████████  ██▀  ▀██      / \   ___ ___(_)___| |_ __ _ _ __ | |_ 
███▀▀███  ██   ▄██     / _ \ / __/ __| / __| __/ _` | '_ \| __|
██    ██  ▀ ▄█████    / ___ \\__ \__ \ \__ \ || (_| | | | | |_ 
███▄▄ ▀█  ▄███████   /_/   \_\___/___/_|___/\__\__,_|_| |_|\__|
▀█████▄   ███████▀

Welcome to the Home Assistant command line.

System information
  IPv4 addresses for wlan0: 
  IPv4 addresses for end0:  192.168.1.3/24
  IPv6 addresses for end0:  fdaf:a30a:3397:740f:f23:456a:c0f9:e6b9/64, fe80::a2af:f2c5:6907:81f7/64

  OS Version:               Home Assistant OS 15.0
  Home Assistant Core:      2025.3.4

  Home Assistant URL:       http://homeassistant.local:8123
  Observer URL:             http://homeassistant.local:4357
➜  ~ wget https://github.com/edwardtfn/TX-Ultimate-Easy/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml 
--2025-03-27 09:47:15--  https://github.com/edwardtfn/TX-Ultimate-Easy/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-03-27 09:47:15 ERROR 404: Not Found.

➜  ~ wget https://github.com/edwardtfn/TX-Ultimate-Easy/blob/main/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml
--2025-03-27 09:47:50--  https://github.com/edwardtfn/TX-Ultimate-Easy/blob/main/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'TX-Ultimate-Easy-ESPHome_core_common.yaml'

TX-Ultimate-Easy-ESPHome_core_commo     [ <=>                                                             ] 356.71K  1.99MB/s    in 0.2s    

2025-03-27 09:47:50 (1.99 MB/s) - 'TX-Ultimate-Easy-ESPHome_core_common.yaml' saved [365276]

➜  ~

chadwo, 2025-03-27:
Using the basic package I get this error:

INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/kitchen-tx.yaml...
Failed config

light.esp32_rmt_led_strip: [source /data/packages/c8868aaa/ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml:47]
  id: light_full
  name: Light - All
  internal: True
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 1
  chipset: ws2812
  num_leds: 32
  pin: GPIO13
  restore_mode: RESTORE_DEFAULT_OFF
  max_refresh_rate: 33ms
  use_psram: True

chadwo, 2025-03-27:
changed to latest release and now compiling with base package. Doing wireless push as already had ESPHome running.

chadwo, 2025-03-30:
Great package!

Having some issues with the backlight coming on unexepectedly. Is there any logic to when it comes on? I have created scripts for the buttons and decoupled the relays. I don't want to disable the lights completely.

edwardtfn, 2025-03-31:
Hey @chadwo, my apologies for the late reply here. Let's take this issue with the backlight on #103.
Back to the original issue when compiling you reported here, is that gone?
Should we close this issue or are you still waiting for some support on that?

chadwo, 2025-04-11:
Sorry for the delay. I am having no further issues with compiling.

closed 2025-05-14 by edwardtfn