edwardtfn/TX-Ultimate-Easy #103: Having some issues with the backlight coming on unexepectedly

edwardtfn, 2025-03-31:
> Great package!
> 
> Having some issues with the backlight coming on unexepectedly. Is there any logic to when it comes on? I have created scripts for the buttons and decoupled the relays. I don't want to disable the lights completely.  

 _Originally posted by @chadwo in [#102](https://github.com/edwardtfn/TX-Ultimate-Easy/issues/102#issuecomment-2764626335)_

edwardtfn, 2025-03-31:
@chadwo, what exactly are you trying to achieve and when is it not behaving as expected?

chadwo, 2025-04-11:
The switch relays are connected to Wiz lights, so the relays should generally stay On at all times. I am using 2 of the 3 relays on the device and have created a toggle action for long press on button 3 only. Pressing buttons 1-3 performs other automations in HA (various Wiz light groupings). However, the "Light" seem unpredictable by default. I created a simple automation to turn it off every time it turns on. I also created an automation to turn the bottom LEDs on Red if one of the Relay 2 is turned off for some reason (just to indicate the relays need to be toggled). All of my Wiz scripts check the relay status and turn on with delay if needed.

I've also noticed the single double and multi-click aren't really predictable enough so my automations just do the same for all actions. This could be just related to my old, fat fingers. Likewise, the swipe actions don't seem very reliable.