edwardtfn/TX-Ultimate-Easy #105: Octal PSRAM is only supported on ESP32-S3, not ESP32.

BartoszPierzchlewiczMacrix, 2025-04-17:
In latest ESPHome 2025.4.0 there is a problem:
`Failed config

psram: [source /data/packages/498351df/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml:111]
  
  Octal PSRAM is only supported on ESP32-S3, not ESP32.
  mode: octal
  speed: 80000000.0
  enable_ecc: False`

ShonP40, 2025-04-30:
Having the same issue

lucidstate, 2025-05-10:
Still getting this error - please help!

error:

INFO ESPHome 2025.4.1
INFO Reading configuration /config/esphome/spare-room-tx-ultimate.yaml...
INFO Updating https://github.com/edwardtfn/TX-Ultimate-Easy@latest
INFO Updating https://github.com/edwardtfn/TX-Ultimate-Easy@v2025.3.1
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

psram: [source /data/packages/498351df/ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml:111]
  
  Octal PSRAM is only supported on ESP32-S3, not ESP32.
  mode: octal
  speed: 80000000.0
  enable_ecc: False

config that is throwing the error:

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: latest  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    refresh: 5min
    files:
      # Core (essential) packages - Do not disable these unless you know what you're doing
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml            # Basic shared settings
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml        # Button logic
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml           # LED configuration
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml          # Touch panel support

      # Optional functionality - Can be enabled/disabled as needed
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml     # Relay control
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml  # Haptic feedback

      # Audio options (use none or choose only one - using both will fail)
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml  # Media player (Recommended for most users)
      # - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_speaker.yaml  # Basic speaker

kylegordon, 2025-05-14:
@lucidstate Use ref: dev in your package declaration instead. Seems to work for me.

ie

`packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: dev  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    refresh: 5min`

edwardtfn, 2025-05-14:
I've just pushed that to `main`, so no need to set `ref: dev`.
Could you please try it and let me know if everything is working fine?

closed 2025-05-14 by edwardtfn

lucidstate, 2025-05-15:
Thanks  Kyle & Edward, everything is working again perfectly :)