edwardtfn/TX-Ultimate-Easy #107: Swipe up & down not available

kylegordon, 2025-05-14:
The Event Documentation suggests there's a swipe up and down event, but it doesn't seem to surface in the list of sensors to enable. Left and right events work (once enabled), but am unable to enable up or down.

I see there's references to bs_swipe_down and bs_swipe_up, but unfortunately I don't understand enough to get a working PR to offer.

edwardtfn, 2025-05-14:
Which model are you using? US or EU?
The EU model supports swipe left/right, while US model supports swipe up/down. This is a hardware limitation.
If you are using a US model and the swipe up/down isn't working, then we have a bug. :(

kylegordon, 2025-05-14:
oh, interesting. I'm using an EU one, all detected automatically. Looks like it's working as expected then. 

I may just rotate the switch in the box then :-D

edwardtfn, 2025-05-14:
That should work, but be aware that it will still reporting as left/right to HA, and the directions will all be 90 degrees off.

closed 2025-05-14 by edwardtfn