edwardtfn/TX-Ultimate-Easy #108: Connectivity issues

ShonP40, 2025-06-05:
When restarting Home Assistant or the access point, the switch can sometimes enter a stuck state where it'll keep functioning (manually) but disconnect from WiFi (not responding to pings) and only reconnect once automatically restarted by the ESPHome API component.

This issue does not happen with any of my other ESPHome devices, including other TX Ultimate (not running `TX Ultimate Easy`) switches.

Home Assistant version: `2025.5.3`
ESPHome version: `2025.5.2`
TX Ultimate Easy version: `2025.5.1`

ESPHome config:

```yaml
substitutions:
  name: "shons-ceiling-lamp"
  friendly_name: "Shon's Ceiling Lamp"

  # Settings Overrides
  BUTTON_CLICK_MIN_LENGTH: '0'    # The minimum duration the click should last, in msec
  BUTTON_CLICK_MAX_LENGTH: '1000'   # The maximum duration the click should last, in msec
  BUTTON_MULTI_CLICK_DELAY: '0'  # The time to wait for another click, in msec
  BUTTON_LONG_PRESS_DELAY: '2000'   # The time to wait to consider a long press, in msec

  LIGHT_TRANSITION_TURN_ON: '500'   # Transition time in msec
  LIGHT_TRANSITION_TURN_OFF: '500'  # Transition time in msec
  LIGHT_BRIGHTNESS_TURN_ON: '50'   # Brightness (1 to 100%, 0 to not set it)

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: latest  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    files:
      # Core (essential) packages - Do not disable these unless you know what you're doing
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml            # Basic shared settings
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml        # Button logic
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml           # LED configuration
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml          # Touch panel support
      # Optional functionality - Can be enabled/disabled as needed
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml     # Relay control
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml  # Haptic feedback
      # Audio options (use none or choose only one - using both will fail)
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml  # Media player (Recommended for most users)
      # - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_speaker.yaml  # Basic speaker

wifi:
  networks:
    ssid: !secret wifi_ssid
    bssid: !secret second_floor_router_bssid
    password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 10.3.204.11
    gateway: 10.0.100.1
    subnet: 255.0.0.0
  ap:
    ssid: ${friendly_name}
    password: !secret wifi_password

captive_portal:

# Enable Home Assistant API
api:
  encryption:
    key: <redacted>

ota:
  platform: esphome
  password: !secret ota_password

web_server:
  port: 80
  version: 3
  auth:
    username: root
    password: !secret wifi_password

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${friendly_name} IP Address
    ssid:
      name: ${friendly_name} SSID
    bssid:
      name: ${friendly_name} BSSID
    mac_address:
      name: ${friendly_name} MAC Address

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime"

  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

switch:
  - platform: factory_reset
    name: ${friendly_name} Restart with Factory Default Settings

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
```

edwardtfn, 2025-06-05:
> disconnect from WiFi (not responding to pings) and only reconnect once automatically restarted by the ESPHome API component

If I understood this correctly, you are still able to restart it from Home Assistant. Is that correct?
If so, your panel still connected to the Wi-Fi, but for some reason (I will investigate this), it stops responding to ping.

ShonP40, 2025-06-05:
No I can't
by API component I meant the `api:` line in ESPHome.

Its not connected to either my WiFi or Home Assistant
Can't ping it & it shows up as unavailable on Home Assistant until I wait 15 minutes for the API component to auto restart the ESP

ShonP40, 2025-06-06:
P.S: I'm using https://github.com/SmartHome-yourself/sonoff-tx-ultimate-for-esphome on my other TX Ultimates (for controlling blinds), which don't have connectivity issues

edwardtfn, 2025-06-07:
Could you please try changing your Wi-Fi settings to `fast_connect: false` and let me know if the same issue persists?

ShonP40, 2025-06-07:
I'm using this setting on all my other ESPs & Sonoff TXs (which run the different config I mentioned above) and they all work just fine
So don't really think that this is the issue...

This setting simply disables WiFi scanning on boot & just tries to connect to the specified BSSID

edwardtfn, 2025-06-07:
I'm out of ideas here. 😩

ShonP40, 2025-06-07:
I'll try to switch to ESP-IDF tomorrow and see if that helps

ShonP40, 2025-06-08:
New config:

```yaml
substitutions:
  name: "shons-ceiling-lamp"
  friendly_name: "Shon's Ceiling Lamp"

  # Settings Overrides
  BUTTON_CLICK_MIN_LENGTH: '0'    # The minimum duration the click should last, in msec
  BUTTON_CLICK_MAX_LENGTH: '1000'   # The maximum duration the click should last, in msec
  BUTTON_MULTI_CLICK_DELAY: '0'  # The time to wait for another click, in msec
  BUTTON_LONG_PRESS_DELAY: '2000'   # The time to wait to consider a long press, in msec

  LIGHT_TRANSITION_TURN_ON: '500'   # Transition time in msec
  LIGHT_TRANSITION_TURN_OFF: '500'  # Transition time in msec
  LIGHT_BRIGHTNESS_TURN_ON: '50'   # Brightness (1 to 100%, 0 to not set it)

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: latest  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    files:
      # Core (essential) packages - Do not disable these unless you know what you're doing
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml            # Basic shared settings
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml        # Button logic
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml           # LED configuration
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml          # Touch panel support
      # Optional functionality - Can be enabled/disabled as needed
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml     # Relay control
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml  # Haptic feedback
      # - ESPHome/TX-Ultimate-Easy-ESPHome_addon_ble_proxy.yaml        # BLE Proxy
      # Audio options (use none or choose only one - using both will fail)
      # - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml  # Media player (Recommended for most users)
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_speaker.yaml  # Basic speaker

wifi:
  networks:
    ssid: !secret wifi_ssid
    bssid: !secret second_floor_router_bssid
    password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 10.3.204.11
    gateway: 10.0.100.1
    subnet: 255.0.0.0
  ap:
    ssid: ${friendly_name}
    password: !secret wifi_password
  power_save_mode: none

captive_portal:

# Enable Home Assistant API
api:
  encryption:
    key: <redacted>

ota:
  platform: esphome
  password: !secret ota_password

web_server:
  port: 80
  version: 3
  auth:
    username: root
    password: !secret wifi_password

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${friendly_name} IP Address
    ssid:
      name: ${friendly_name} SSID
    bssid:
      name: ${friendly_name} BSSID
    mac_address:
      name: ${friendly_name} MAC Address

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime"

  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

switch:
  - platform: factory_reset
    name: ${friendly_name} Restart with Factory Default Settings

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
#########################################################
esp32:
  framework:
    type: esp-idf

light:
  - id: !extend light_full
    platform: esp32_rmt_led_strip
    rgb_order: GRB
    chipset: ws2812
    max_refresh_rate: 33ms
    use_psram: false
    type: !remove
    variant: !remove
```

It seems a bit more stable, but only time will tell

(also, configuring speaker feedback doesn't make any sounds)

ShonP40, 2025-06-14:
Seems to be stable on the one switch I installed this on, so gonna install it on my other 2 switches and report back in a few days if its stable or not

ShonP40, 2025-06-20:
Seems to be 100% stable now
I'm not sure what exactly fixed it though. Could've been ditching the Arduino framework, or just manually erasing & flashing new firmware.

I'll be staying on ESP-IDF just to be safe

closed 2025-06-20 by ShonP40