edwardtfn/TX-Ultimate-Easy #114: Request for Review: Automation Logic Integration with TX Ultimate Easy

mikesn84, 2025-07-22:
Hello,

I am using your [TX Ultimate Easy ESPHome project](https://github.com/edwardtfn/TX-Ultimate-Easy) on a Sonoff T5-2C-86 and have integrated a time-based automation to control a pool filter pump (relay 1) and salt system (relay 2).

My goal is:

To have a web interface where I can set filter run times and total filter duration in hours.
A virtual switch in Home Assistant to enable/disable the automation.
When the automation is ON, relay 1 (filter pump) and relay 2 (salt system) should turn ON/OFF automatically based on the schedule.
When automation is OFF, manual control of relays should be possible.
The **salt system** relay (relay 2) must always turn OFF whenever the filter pump relay (relay 1) turns OFF to prevent dry running.
I adapted my ESPHome YAML to load your package as a remote package and added these features.

Could you please review my approach and let me know if this integration looks correct or if there are any pitfalls I should be aware of?

```
substitutions:
  name: tx-ultimate-easy
  friendly_name: TX Ultimate Easy

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: stable
    refresh: 5min
    files:
      - ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard.yaml

# Fallback Hotspot, falls WLAN nicht geht
ap:
  ssid: "TX-UltimateEasy-Setup"
  password: "12345678"

# Webserver aktivieren, Port 80, mit internem UI
web_server:
  port: 80
  include_internal: true

# Virtueller Schalter für Automatik EIN/AUS
switch:
  - platform: template
    name: "Automatik Poolfilter"
    id: switch_automatik
    optimistic: true
    restore_state: true

# Textfeld und Zahlfeld für Zeiten und Filterdauer
text:
  - platform: template
    name: "Startzeiten (HH:MM;HH:MM;...)"
    id: startzeiten_text
    optimistic: true
    restore_state: true

number:
  - platform: template
    name: "Filterdauer in Stunden"
    id: filterdauer_stunden
    min_value: 0.5
    max_value: 24
    step: 0.1
    optimistic: true
    restore_state: true

# Automatik-Timer (alle 5 Minuten)
interval:
  - interval: 300s
    then:
      - lambda: |-
          if (!id(switch_automatik).state) {
            // Automatik aus -> nichts tun
            return;
          }
          // Zeiten einlesen
          std::string raw_times = id(startzeiten_text).state.c_str();
          float dauer_gesamt = id(filterdauer_stunden).state;
          if(raw_times.length() == 0 || dauer_gesamt <= 0) {
            return;  // Keine Zeiten gesetzt oder Dauer 0
          }
          
          // Zeiten aufsplitten
          std::vector<std::string> times;
          std::stringstream ss(raw_times);
          std::string item;
          while (std::getline(ss, item, ';')) {
            times.push_back(item);
          }

          // aktuelle Zeit in Minuten seit Mitternacht
          auto now = id(homeassistant_time).now();
          int jetzt_min = now.hour * 60 + now.minute;

          int anzahl = times.size();
          int laufzeit_min = (int)((dauer_gesamt / anzahl) * 60);

          bool aktiv = false;

          for(auto &zeit : times) {
            int h = 0, m = 0;
            if (sscanf(zeit.c_str(), "%d:%d", &h, &m) == 2) {
              int start_min = h * 60 + m;
              int stop_min = (start_min + laufzeit_min) % (24 * 60);
              // einfache Prüfung auf aktiv (Über Mitternacht nicht behandelt)
              if(jetzt_min >= start_min && jetzt_min < stop_min) {
                aktiv = true;
                break;
              }
            }
          }

          if(aktiv) {
            // Wenn Filter aus -> einschalten
            if(!id(relay_1).state) {
              id(relay_1).turn_on();
              id(relay_2).turn_on(); // Salzanlage mit einschalten
              ESP_LOGD("automatik", "Filter und Salzanlage eingeschaltet");
            }
          } else {
            // Wenn Filter an -> ausschalten
            if(id(relay_1).state) {
              id(relay_1).turn_off();
              id(relay_2).turn_off(); // Salzanlage mit ausschalten
              ESP_LOGD("automatik", "Filter und Salzanlage ausgeschaltet");
            }
          }

# Referenz zum aktuellen Zeitobjekt von HA
time:
  - platform: homeassistant
    id: homeassistant_time
```

Thank you very much for your help and your great work on this project!

Best regards,
Mike

edwardtfn, 2025-07-22:
I will look at your `interval` logic later when I have access from my computer, but for now, l would suggest some protection to prevent relay 2 getting on without relay 1 being on. Something like this shoukd be sufficient if you can accept relay 2 turning on a few milliseconds before relay 1 (otherwise we can elaborate a bit more):

```yaml
switch:
  - id: !extend relay_1
    on_turn_off:
      then:
        - switch.turn_off: relay_2
  - id: !extend relay_2
    on_turn_on:
      then:
        - switch.turn_on: relay_1
```
With this, relay 2 will turn off automatically when you turn off relay 1 and relay 1 will turn on automatically when you turn on relay 2.
This will prevent mistakes when controlling it manually from Home Assistant side.

edwardtfn, 2025-07-22:
By the way, have you tried replacing your timer logic entirely by a delay directly on the relay?
Something like this:
```yaml
switch:
  - id: !extend relay_1
    on_turn_on:
      then:
        - delay: !lamba return id(filterdauer_stunden)*60*60*1000;
        - switch.turn_off: relay_2
        - switch.turn_off: relay_1
```

edwardtfn, 2025-07-31:
What about this? (Not tested - just for reference)
```yaml
####################################################################################################

##### TX Ultimate Easy - Pool Filter Automation

##### Based on TX Ultimate Easy for ESPHome

## ####################################################################################################

substitutions:
name: tx-ultimate-easy
friendly_name: TX Ultimate Easy

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password

# Fallback hotspot with German naming convention

ap:
ssid: “TX-UltimateEasy-Setup”
password: “12345678”

packages:
remote_package:
url: https://github.com/edwardtfn/TX-Ultimate-Easy
ref: stable
refresh: 5min
files:
- ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml
- ESPHome/TX-Ultimate-Easy-ESPHome_standard.yaml

# Web server configuration - moved outside packages to avoid conflicts

web_server:
port: 80
include_internal: true
auth:
username: !secret web_server_username
password: !secret web_server_password

# Pool automation components

globals:

# Global flag to prevent rapid switching

- id: automation_running
  type: bool
  restore_value: false
  initial_value: ‘false’

# Last state change timestamp to prevent oscillation

- id: last_state_change
  type: uint32_t
  restore_value: false
  initial_value: ‘0’

# Current active cycle information (persistent across reboots)

- id: current_cycle_start_time
  type: uint32_t
  restore_value: true
  initial_value: ‘0’
- id: current_cycle_duration
  type: uint32_t
  restore_value: true
  initial_value: ‘0’
[... 340 lines cut ...]
  
  auto time_now = id(homeassistant_time).now();
  if (!time_now.is_valid()) {
  return {};
  }
  
  uint32_t current_time_minutes = time_now.hour * 60 + time_now.minute;
  uint32_t cycle_start_minutes = id(current_cycle_start_time);
  uint32_t cycle_duration_minutes = id(current_cycle_duration);
  
  uint32_t elapsed_minutes = 0;
  if (current_time_minutes >= cycle_start_minutes) {
  elapsed_minutes = current_time_minutes - cycle_start_minutes;
  } else {
  // Crossed midnight
  elapsed_minutes = (24 * 60 - cycle_start_minutes) + current_time_minutes;
  }
  
  if (elapsed_minutes >= cycle_duration_minutes) {
  return 0;  // Cycle should be finished
  }
  
  return (float)(cycle_duration_minutes - elapsed_minutes);
  update_interval: 60s
```

closed 2025-08-05 by edwardtfn