edwardtfn/TX-Ultimate-Easy #124: Device keeps rebooting when enabling Bluetooth Proxy

mon3y78, 2025-09-09:
Hi, first of all thank you for your amazing work on this project — I really appreciate the effort and the quality of what you’ve built.

I’m facing an issue: whenever I enable the Bluetooth Proxy, my device keeps rebooting continuously.
I’m attaching my configuration and the relevant logs below to help with troubleshooting.

Configuration:
---
substitutions:
  name: "tx-salotto"
  friendly_name: TX Ultimate salotto

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: v2025.8.2  # Or you can speci8fy a version, like `ref: v2024.12.6` or `ref: latest`
    refresh: 5min
    files:
      # Core (essential) packages
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_common.yaml            # Basic shared settings
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_buttons.yaml        # Button logic
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_leds.yaml           # LED configuration
      - ESPHome/TX-Ultimate-Easy-ESPHome_core_hw_touch.yaml          # Touch panel support

      # Optional but recommended packages
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_relays.yaml     # Relay control
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_vibration.yaml  # Haptic feedback

      # Audio options (use none or choose only one - using both will fail)
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard_media_player.yaml  # Media player (Recommended for most users)
      # - ESPHome/TX-Ultimate-Easy-ESPHome_standard_hw_speaker.yaml  # Basic speaker

# Optional: Add Bluetooth proxy support
bluetooth_proxy:

log:
[logs (7).txt](https://github.com/user-attachments/files/22228134/logs.7.txt)

edwardtfn, 2025-12-10:
Could you please try this and let me know if you get success?

```yaml
substitutions:
  name: "front_hall_dual_switch"
  friendly_name: "Front Hall Dual Switch"
  device_description: "Front Hall Dual Light Switch"
  default_state: "RESTORE_DEFAULT_ON"

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: main  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    refresh: 5min
    files:
      - ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml      # Core (essential) packages
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard.yaml  # Non-essential, but recommended packages. Includes hw_relays, hw_vibration, and media_player

bluetooth_proxy:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${friendly_name} AP
    password: !secret wifi_password
  use_psram: true

esp32:
  framework:
    advanced:
      loop_task_stack_size: 16384
```

Basically I'm adding the use of PSRAM for Wi-Fi and increasing the loop stack.

kylegordon, 2025-12-10:
@edwardtfn I think I spoke too soon, it's crashing using your config too now. The only way I can get it to work reliably is to remove the bluetooth_proxy line, even with your config. I don't know what's changed tbh.

I've attached  some logs of it crashing, if that's of any use.

[txultimateeasy.log](https://github.com/user-attachments/files/24086791/txultimateeasy.log)

edwardtfn, 2025-12-10:
Well, I can work in the code to make it leaner... transferring the selection between EU/US or number of gangs from UI to yaml would make a huge difference on the memory usage, but will be less "Easy"... :(
I can look at that, but not before the holidays. 😞 
[ESPHome PR#12213](https://github.com/esphome/esphome/pull/12213) could also help making it possible via yaml while keeping it quite easy, But that wasn't included on ESPHome v2025.12.0-BETA, so we will have to wait at least one more month.

edwardtfn, 2025-12-10:
> [@edwardtfn](https://github.com/edwardtfn) I think I spoke too soon, it's crashing using your config too now. The only way I can get it to work reliably is to remove the bluetooth_proxy line, even [...]

Can you try getting a [Stack trace](https://esphome.io/guides/troubleshooting/#getting-a-stack-trace-from-crashes)?

kylegordon, 2025-12-12:
Sadly it's still crashing, but only when bluetooth_proxy: is enabled. 

Log attached, config file below, just in case I've missed a line or something

```
substitutions:
  name: "front_hall_dual_switch"
  friendly_name: "Front Hall Dual Switch"
  device_description: "Front Hall Dual Light Switch"
  default_state: "RESTORE_DEFAULT_ON"

packages:
  remote_package:
    url: https://github.com/edwardtfn/TX-Ultimate-Easy
    ref: main  # Or you can specify a version, like `ref: v2024.12.6` or `ref: latest` to the latest non-stable
    refresh: 5min
    files:
      - ESPHome/TX-Ultimate-Easy-ESPHome_core.yaml      # Core (essential) packages
      - ESPHome/TX-Ultimate-Easy-ESPHome_standard.yaml  # Non-essential, but recommended packages. Includes hw_relays, hw_vibration, and media_player

external_components:
  - source:
      type: git
      url: https://github.com/edwardtfn/TX-Ultimate-Easy
      ref: v9999.99.9
    refresh: 1min
    components:
      - tx_ultimate_easy

bluetooth_proxy:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${friendly_name} AP
    password: !secret wifi_password
  use_psram: true

esp32:
  framework:
    advanced:
      loop_task_stack_size: 16384

```

[txultimateeasy.log](https://github.com/user-attachments/files/24128369/txultimateeasy.log)

kylegordon, 2025-12-27:
All working great so far 👍

kaantursucu, 2026-01-21:
I got 19  from 1 gang to 4 i will try it again let you know for 2 -3-4 gang